f4pga/prjxray #1913: New GitHub actions based CI does not provide Vivado license

hansfbaier, 2022-04-22:
This pull request runs successfully on my local machine,
but fails in the CI, because the added part is not available in the WebPack edition:
https://github.com/f4pga/prjxray/pull/1908

mithro, 2022-04-24:
@acomodi / @kgugala - Can you take a look?

acomodi, 2022-04-25:
@mithro One necessary step to solve this will most probably be the addition of the [tunnel key](https://github.com/f4pga/prjxray/blob/master/.github/kokoro/steps/xilinx.sh#L64) as a GH secret, so that we can access the license server. We do not have access to the `$KOKORO_KEYSTORE_DIR` though, and I am not quite sure where to get it. I see it is currently fetched before running the kokoro action: https://github.com/f4pga/prjxray/blob/master/.github/kokoro/continuous-db-kintex7.cfg#L46-L53

mithro, 2022-04-25:
@acomodi - We can set this up temporarily, but we should set it up so that the GitHub Runners on GCP can login to the tunnel server without needing a SSH key. I think there is some documentation at https://cloud.google.com/compute/docs/tutorials/service-account-ssh ?

mithro, 2022-04-25:
I have created the `LICENSE_TUNNEL_KEY_DATA` GitHub secret. The contents of this file should be dumped into the `LICENSE_TUNNEL_KEY` file.

mithro, 2022-04-25:
Testing in the branch at https://github.com/f4pga/prjxray/tree/xilinx

mithro, 2022-04-25:
@acomodi - It seems like the commands are being run as root?

mithro, 2022-04-25:
I think what is needed is now shown in https://github.com/f4pga/prjxray/pull/1918/files

We just need to make sure that the Google Cloud Platform which has the runners can connect to the Xilinx license server machine.

mithro, 2022-04-25:
Okay, I think we have xilinx-license servers setup in the correct GCP projects, however I'm unclear how to the runners can access the GCP private network (`10.138.0.0`).

The changes can be found here -> https://github.com/f4pga/prjxray/pull/1918/files

See the output at https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true
```
Run .github/workflows/scripts/xilinx.sh
23:23:07 | 
23:23:07 | ========================================
23:23:07 | Xilinx proprietary toolchain setup.
23:23:07 | ----------------------------------------
23:23:07 | 
23:23:07 | Fix up the Xilinx configuration directory
23:23:07 | ----------------------------------------
23:23:07 | total 0
23:23:07 | ----------------------------------------
23:23:07 | 
23:23:07 | Select Xilinx Vivado version
23:23:07 | ----------------------------------------
23:23:07 | Using Xilinx Vivado WebPack Edition for  build.
23:23:07 | ls: cannot access 'Xilinx': No such file or directory
23:23:07 | ----------------------------------------
23:23:07 | 
23:23:07 | List /opt directory
23:23:07 | ----------------------------------------
23:23:07 | total 0
23:23:07 | ----------------------------------------
23:23:07 | 
23:23:07 | Setting up license server tunnel
23:23:07 | ----------------------------------------
23:23:07 | SSH Key for license server tunnel should be found @ /root/.Xilinx/xilinx-tunnel-key
23:23:07 | -rw-r--r-- 1 root root 2610 Apr 25 23:23 /root/.Xilinx/xilinx-tunnel-key
23:23:07 | 
23:23:07 | Xilinx license server ssh key found, setting up tunnel
23:23:07 | 
23:23:07 | 
23:23:07 | Checking 10.138.0.2 is up
23:23:07 | 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
23:23:07 |     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
23:23:07 |     inet 127.0.0.1/8 scope host lo
23:23:07 |        valid_lft forever preferred_lft forever
23:23:07 |     inet6 ::1/128 scope host 
23:23:07 |        valid_lft forever preferred_lft forever
23:23:07 | 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
23:23:07 |     link/ether 42:01:0a:00:00:4a brd ff:ff:ff:ff:ff:ff
23:23:07 |     inet 10.0.0.74/32 brd 10.0.0.74 scope global dynamic eth0
23:23:07 |        valid_lft 86332sec preferred_lft 75532sec
23:23:07 |     inet6 fe80::d406:321b:ce8e:21[9](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:9)b/64 scope link 
23:23:07 |        valid_lft forever preferred_lft forever
23:23:07 | ping: ---------
23:23:22 | PING 10.138.0.2 (10.138.0.2) 56(84) bytes of data.
23:23:22 | 
23:23:22 | --- [10](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:10).[13](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:13)8.0.2 ping statistics ---
23:23:22 | 5 packets transmitted, 0 received, 100% packet loss, time 4081ms
23:23:22 | 
23:23:22 | port: ---------
23:25:32 | nc: connect to 10.138.0.2 port 22 (tcp) failed: Connection timed out
23:25:32 | ---------
23:25:32 | 
23:25:32 | .github/workflows/scripts/xilinx.sh: line 98: sudo: command not found
23:27:43 | ssh: connect to host 10.138.0.2 port 22: Connection timed out
23:27:43 | .github/workflows/scripts/xilinx.sh: line 104: /opt/Xilinx/Vivado/20[17](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:17).2/settings64.sh: No such file or directory
[23](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:23):[27](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:27):[43](https://github.com/f4pga/prjxray/runs/6166925335?check_suite_focus=true#step:5:43) | -----
23:27:43 | .github/workflows/scripts/xilinx.sh: line 107: lmutil: command not found
23:27:43 | -----
23:27:43 | ----------------------------------------
```

@kgugala / @pgielda -- I think I need your assistance here to fix this now....

closed 2022-10-25 by kgugala