f4pga/prjxray #1950: Incompatible with GCC >= 11 (default -std=c++17)

Xiretza, 2022-05-29:
GCC 11 [updated](https://gcc.gnu.org/gcc-11/changes.html) the default `-std=` for C++ to `gnu++17`. This causes abseil (which specifies no explicit `-std=`) to be built as C++17, while the rest of prjxray is build with `c++14`: https://github.com/f4pga/prjxray/blob/master/CMakeLists.txt#L40

This is explicitly unsupported by abseil: https://github.com/abseil/abseil-cpp/blob/48f72c227b94b06387106f71d4450b31e88e283b/absl/base/options.h#L144-L151

It causes the following linker error:

```
[...]
[ 80%] Linking CXX executable bits2rbt
cd /build/prjxray-git/src/build/tools/bits2rbt && /usr/bin/cmake -E cmake_link_script CMakeFiles/bits2rbt.dir/link.txt --verbose=1
/usr/bin/c++ -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -g -fvar-tracking-assignments -ffile-prefix-map=/build/prjxray-git/src=/usr/src/debug -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now CMakeFiles/bits2rbt.dir/bits2rbt.cc.o CMakeFiles/bits2rbt.dir/configuration_packets.cc.o CMakeFiles/bits2rbt.dir/header.cc.o CMakeFiles/bits2rbt.dir/ecc.cc.o -o bits2rbt  ../../third_party/abseil-cpp/absl/strings/libabsl_strings.a ../../third_party/gflags/libgflags_nothreads.a ../../third_party/abseil-cpp/absl/strings/libabsl_strings_internal.a ../../third_party/abseil-cpp/absl/base/libabsl_base.a ../../third_party/abseil-cpp/absl/base/libabsl_spinlock_wait.a -lrt ../../third_party/abseil-cpp/absl/numeric/libabsl_int128.a ../../third_party/abseil-cpp/absl/base/libabsl_throw_delegate.a ../../third_party/abseil-cpp/absl/base/libabsl_raw_logging_internal.a ../../third_party/abseil-cpp/absl/base/libabsl_log_severity.a 
/usr/bin/ld: CMakeFiles/bits2rbt.dir/configuration_packets.cc.o: in function `absl::strings_internal::Splitter<absl::strings_internal::SelectDelimiter<char const*>::type, absl::AllowEmpty, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > absl::StrSplit<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, 0>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, char const*)':
/usr/src/debug/prjxray/third_party/abseil-cpp/absl/strings/str_split.h:516: undefined reference to `absl::ByString::ByString(absl::string_view)'
/usr/bin/ld: CMakeFiles/bits2rbt.dir/configuration_packets.cc.o: in function `absl::strings_internal::SplitIterator<absl::strings_internal::Splitter<absl::ByString, absl::AllowEmpty, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::operator++()':
/usr/src/debug/prjxray/third_party/abseil-cpp/absl/strings/internal/str_split_internal.h:133: undefined reference to `absl::ByString::Find(absl::string_view, unsigned long) const'
/usr/bin/ld: /usr/src/debug/prjxray/third_party/abseil-cpp/absl/strings/internal/str_split_internal.h:133: undefined reference to `absl::ByString::Find(absl::string_view, unsigned long) const'
/usr/bin/ld: CMakeFiles/bits2rbt.dir/header.cc.o: in function `absl::string_view::find(char const*, unsigned long) const':
/usr/src/debug/prjxray/third_party/abseil-cpp/absl/strings/string_view.h:473: undefined reference to `absl::string_view::find(absl::string_view, unsigned long) const'
/usr/bin/ld: /usr/src/debug/prjxray/third_party/abseil-cpp/absl/strings/string_view.h:473: undefined reference to `absl::string_view::find(absl::string_view, unsigned long) const'
```

...because abseil compiles using `std::string_view`, while `bits2rbt` links against `absl::string_view`.

LAK132, 2022-08-16:
This change to the root CMakeLists.txt seems to fix this: 
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 3ec89c6b..5fa4b525 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -3,6 +3,8 @@ cmake_minimum_required(VERSION 3.5.0)
 project(prjxray)
 option(PRJXRAY_BUILD_TESTING "" OFF)
 
+set(CMAKE_CXX_STANDARD 14)
+
 # Add sanitizers-cmake package
 set(CMAKE_MODULE_PATH "${CMAKE_SOURCE_DIR}/third_party/sanitizers-cmake/cmake" ${CMAKE_MODULE_PATH})
 find_package(Sanitizers)
@@ -36,8 +38,6 @@ target_include_directories(yaml-cpp PUBLIC
 	$<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/third_party/yaml-cpp/include>
 	)
 
-# Set the CXX standard and compile time for our code only.
-set(CMAKE_CXX_STANDARD 14)
 add_compile_options(-Wall -Werror)
 
 add_subdirectory(lib)
```

calvinboey, 2022-08-27:
Thanks @LAK132 I confirm that the changes to CMakeLists.txt that you proposed fixed the following issue I had

`[ 88%] Linking CXX executable bitread
/usr/bin/ld: CMakeFiles/bitread.dir/bitread.cc.o: in function `main':
bitread.cc:(.text.startup+0xac9): undefined reference to `absl::ByString::ByString(absl::string_view)'
/usr/bin/ld: bitread.cc:(.text.startup+0x2ee6): undefined reference to `absl::variant_internal::ThrowBadVariantAccess()'
/usr/bin/ld: CMakeFiles/bitread.dir/bitread.cc.o: in function `_ZNK4absl16strings_internal8SplitterINS_8ByStringENS_10AllowEmptyENS_11string_viewEEcvSt4pairIT_T0_EINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_EEv':
bitread.cc:(.text._ZNK4absl16strings_internal8SplitterINS_8ByStringENS_10AllowEmptyENS_11string_viewEEcvSt4pairIT_T0_EINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_EEv[_ZNK4absl16strings_internal8SplitterINS_8ByStringENS_10AllowEmptyENS_11string_viewEEcvSt4pairIT_T0_EINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_EEv]+0xe3): undefined reference to `absl::ByString::Find(absl::string_view, unsigned long) const'`

tmichalak, 2023-03-17:
Fixed with #2118. Closing.

closed 2023-03-17 by tmichalak