f4pga/prjxray #2050: Compilation error

zipotron, 2022-11-17:
Hello, I am trying to compile last version of the repo and I found a compilation error. I suspect there is a regression because one month ago I did compile successfully in another computer:
[image]

Matzikratzi, 2022-11-20:
I get the same error. I try compile on Ubuntu 22.10. Could it be that you used another distribution/version last time when it worked?
I do not have any computers with another OS, so I cannot check.

Matzikratzi, 2022-11-20:
Probably duplicate to #1950

Matzikratzi, 2022-11-20:
Yes, 'make build' works on Ubuntu 22.10 with this fix.
gcc 12.2.0-3ubuntu1

zipotron, 2022-11-21:
Right! Is fix the issue! Thanks

closed 2022-11-21 by zipotron