f4pga/prjxray #2127: struct std::iterator’ is deprecated in segbits_file_reader.h:38

buggbear, 2023-03-20:
When compiling with cmake 3.25.1 I get this error:

/home/xyz/prjxray/lib/include/prjxray/segbits_file_reader.h:38:27: error: ‘template<class _Category, class _Tp, class _Distance, class _Pointer, class _Reference> struct std::iterator’ is deprecated [-Werror=deprecated-declarations]
   38 |             : public std::iterator<std::input_iterator_tag, value_type> {
      |                           ^~~~~~~~
In file included from /usr/include/c++/12/bits/stl_algobase.h:65,
                 from /usr/include/c++/12/memory:63,
                 from /home/xyz/prjxray/lib/include/prjxray/database.h:13:
/usr/include/c++/12/bits/stl_iterator_base_types.h:127:34: note: declared here
  127 |     struct _GLIBCXX17_DEPRECATED iterator
      |                                  ^~~~~~~~
cc1plus: all warnings being treated as errors
make[3]: *** [lib/CMakeFiles/libprjxray.dir/build.make:76: lib/CMakeFiles/libprjxray.dir/database.cc.o] Error 1

shake-coder, 2023-03-25:
This error message indicates that the code is using the std::iterator class, which has been marked as deprecated in the C++17 standard.

The deprecated warning is being treated as an error, which is causing the compilation to fail. A quick fix would be to remove the `-Werror` flag from `add_compile_options(-Wall -Werror)` in "prjxray/CMakeLists.txt'. The deprecated warning will no longer be treated as an error, and the compilation will not fail. Instead, the warning will be displayed as a message during compilation, and the build process will continue. 

For a better long-term solution, the code needs to updated to use replacements for deprecated features.

shake-coder, 2023-03-27:
#2135

aceOrange, 2023-05-07:
Met the same issue still, but was able to work around it by fixing the c++ version to 14, in the Cmakelists.txt file, by adding

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)