f4pga/prjxray #2169: Compilation error "make build"

zipotron, 2023-05-22:
Hi everybody, I am using a recent updated Fedora 38, I was compiling this project in Fedora 36 without issues, but now I get this error:
[image]

Any suggestion? Thanks!

Amjad2004, 2023-05-22:
>
> Password reset
>
User ID Amjad2004
Gmail id ***@***.***

> Message ID: ***@***.***>
>

tmichalak, 2023-05-25:
@zipotron the problem you have is related to the fact that the code is using the std::iterator class, which has been marked as deprecated in the C++17 standard.
There is a [PR](#2135) with a fix.

pgielda, 2023-05-26:
The PR mentioned above (https://github.com/f4pga/prjxray/pull/2135) is merged.

zipotron, 2023-06-01:
Works! Thanks!

closed 2023-06-01 by zipotron