f4pga/prjxray #2300: prjxray-db inconsistend, bits resulting in invalid word addresses (>100)

hansfbaier, 2024-01-08:
It looks like the prjxray-db is inconsistent, for example in Artix7/Spartan7 there is
```
    "LIOI3_SING_X0Y149": {
        "bits": {
            "CLB_IO_CLK": {
                "alias": {
                    "sites": {},
                    "start_offset": 0,
                    "type": "LIOI3"
                },
                "baseaddr": "0x00020000",
                "frames": 42,
                "offset": 99,
                "words": 2
            }
        },
```
Then there are features like:
```
segbits_lioi3.db:LIOI3.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
segbits_lioi3_tbytesrc.db:LIOI3_TBYTESRC.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
segbits_lioi3_tbyteterm.db:LIOI3_TBYTETERM.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
segbits_rioi3.db:RIOI3.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
segbits_rioi3_tbytesrc.db:RIOI3_TBYTESRC.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
segbits_rioi3_tbyteterm.db:RIOI3_TBYTETERM.IOI_ILOGIC0_CLK.IOI_LEAF_GCLK0 28_67 28_79 29_74
```
When these feature are used in the SING tile above, then the corresponding word address is 101:
The word address is calculated:
```
word_addr = bit.word_bit // bitstream.WORD_SIZE_BITS
```
So in this example bit 28_79 gives word address 99 + 79//32  = 99+2 = 101.
But each frame only has 101 words, with word addresses ranging from 0..100
So either the offset is wrong in tilegrid.json, or the bits in those features are wrong.

hansfbaier, 2024-01-11:
The reason for this is that SING tiles have different switchboxes and their own set of PIPs, which are currently not supported by prjxray
[image]

gHashTag, 2026-10-02:
A data point on the `*_SING` alias question, from a check that does not go through FASM.

I took the four Vivado bitstreams in `prjxray-db/artix7/harness` (arty-a7 swbut, uart and pmod, and basys3 swbut; all xc7a35t, read against prjxray-db 0a0adde, `xc7a50t/tilegrid.json`). For every set bit, I asked whether some tile's segbits name that bit at the frame and word the FAR walk assigns to the tile.

| harness | named by the tile's own segbits | named once the tilegrid `alias` (with `start_offset`) is applied |
|---|---|---|
| arty-a7/swbut | 796 / 809 | 809 / 809 |
| arty-a7/uart | 255 / 255 | 255 / 255 |
| arty-a7/pmod | 881 / 890 | 890 / 890 |
| basys3/swbut | 1831 / 1844 | 1844 / 1844 |

The 13, 0, 9 and 13 bits that only the alias names all sit in `*_SING` IO tiles (RIOB33/RIOI3_SING_X43Y50, LIOB33/LIOI3_SING_X0Y99, LIOB33/LIOI3_SING_X0Y0).

What this does not cover: none of these designs uses a top `*_SING` tile, so it says nothing about the word > 100 case in this issue. For that case, openXC7/prjxray#20 and #21 (merged 2026-09-18) set `start_offset` 0 for the top-SING IOB33/IOI3 and IOB18/IOI tiles. If someone has a Vivado bitstream that drives a pin in a top SING tile, the same check would test that fix directly.

🤖 Generated with [Claude Code](https://claude.com/claude-code)